ros2/kilted_tutorial_party #75: Check ROS 2 run CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check run**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You'll have to kill this command with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
       [INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
       [INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
       ...
       ```
  </details>
- [ ] **Check run with prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-03:
I'll test this on Windows 11 / amd64 / binary / Fastdds

ShahazadAbdulla, 2025-05-03:
Completed checks for Issue https://github.com/ros2/kilted_tutorial_party/issues/73 on Windows 11 / amd64 / binary / FastDDS.

Check 1 (ros2 run demo_nodes_cpp talker): Passed. Node started and published messages as expected. Pressing Ctrl+C stopped it normally.

Check 2 (ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker): Could not perform check. gdb is not found in the standard Windows binary installation environment for Kilted.

Additional Test (ros2 run --prefix "start" demo_nodes_cpp talker): To test the prefix mechanism itself, I tried using the built-in start command. This FAILED with a FileNotFoundError: [WinError 2] The system cannot find the file specified. Traceback below. This might indicate an issue with how --prefix constructs or executes commands on Windows.

Traceback for --prefix "start":

Traceback (most recent call last):
File "\?\C:\pixi_ws\ros2-windows\Scripts\ros2-script.py", line 33, in <module>
sys.exit(load_entry_point('ros2cli==0.38.0', 'console_scripts', 'ros2')())
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\pixi_ws\ros2-windows\Lib\site-packages\ros2cli\cli.py", line 91, in main
rc = extension.main(parser=parser, args=args)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\pixi_ws\ros2-windows\Lib\site-packages\ros2run\command\run.py", line 70, in main
return run_executable(path=path, argv=args.argv, prefix=prefix)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\pixi_ws\ros2-windows\Lib\site-packages\ros2run\api_init_.py", line 64, in run_executable
process = subprocess.Popen(cmd)
^^^^^^^^^^^^^^^^^^^^^
File "C:\pixi_ws.pixi\envs\default\Lib\subprocess.py", line 1026, in init
self._execute_child(args, executable, preexec_fn, close_fds,
File "C:\pixi_ws.pixi\envs\default\Lib\subprocess.py", line 1538, in _execute_child
hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified
Summary: Basic ros2 run works, but the --prefix 'gdb ...' check is not feasible on the binary install, and using --prefix "start" resulted in an error.

mjcarroll, 2025-05-09:
Yes, the `--prefix gdb` probably makes less sense on Windows, noted here: https://github.com/osrf/ros2_test_cases/issues/1694

closed 2025-05-09 by mjcarroll