ros2/kilted_tutorial_party #77: Check ROS 2 run CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check run**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You'll have to kill this command with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
       [INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
       [INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
       ...
       ```
  </details>
- [ ] **Check run with prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
```
WARN[0000] /home/suchetan/kilted_tnt/ros2_kilted_deb/gz-sim-environment/docker-compose.yml: `version` is obsolete 
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_zenoh_cpp
root@cluttered:~# ros2 run rmw_zenoh_cpp rmw_zenohd &
[1] 1345281
root@cluttered:~# 2025-05-02T19:57:41.559372Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 1436fc131702b3677ff882f11f2c1ecc
2025-05-02T19:57:41.560706Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2405:201:d000:d170:b306:a034:1275:5935]:7447
2025-05-02T19:57:41.560731Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2405:201:d000:d170:a1be:4501:9190:2bc9]:7447
2025-05-02T19:57:41.560739Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::a901:2c0:950f:8242]:7447
2025-05-02T19:57:41.560747Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.29.137:7447
Started Zenoh router with id 1436fc131702b3677ff882f11f2c1ecc
^C
root@cluttered:~# # User input in terminal 1
ros2 run demo_nodes_cpp talker
[INFO] [1746215869.708469102] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746215870.708478899] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746215871.708484725] [talker]: Publishing: 'Hello World: 3'
^C[INFO] [1746215872.452748740] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746215872.452998083] [rclcpp]: signal_handler(signum=2)
```

With GDB: (Had to install gdb manually, it wasn't a part of desktop-full)

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_zenoh_cpp
root@cluttered:~# # User input in terminal 1
ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kilted/lib/demo_nodes_cpp/talker...
(No debugging symbols found in /opt/ros/kilted/lib/demo_nodes_cpp/talker)
Starting program: /opt/ros/kilted/lib/demo_nodes_cpp/talker 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/liblttng-ust.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/liblttng-ust-tracepoint.so.1
[New Thread 0x7ffff740e6c0 (LWP 1348589)]
[New Thread 0x7ffff6c0d6c0 (LWP 1348590)]
[New Thread 0x7ffff58806c0 (LWP 1348591)]
[New Thread 0x7ffff567f6c0 (LWP 1348592)]
[New Thread 0x7ffff547e6c0 (LWP 1348593)]
[New Thread 0x7ffff527d6c0 (LWP 1348594)]
[New Thread 0x7ffff507c6c0 (LWP 1348595)]
[New Thread 0x7ffff4e7b6c0 (LWP 1348596)]
[New Thread 0x7ffff4c7a6c0 (LWP 1348597)]
[New Thread 0x7ffff4a796c0 (LWP 1348598)]
[New Thread 0x7ffff48786c0 (LWP 1348599)]
[New Thread 0x7ffff46776c0 (LWP 1348600)]
[New Thread 0x7fffcffff6c0 (LWP 1348601)]
[New Thread 0x7fffcf7fe6c0 (LWP 1348602)]
[INFO] [1746215983.449100122] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746215984.449066085] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746215985.449049763] [talker]: Publishing: 'Hello World: 3'
^C
Thread 1 "talker" received signal SIGINT, Interrupt.
0x00007ffff790ed71 in __futex_abstimed_wait_common64 (private=1262903851, cancel=true, abstime=0x7fffffffbd80, op=137, expected=0, futex_word=0x5555555bf128)
    at ./nptl/futex-internal.c:57
warning: 57	./nptl/futex-internal.c: No such file or directory
(gdb) q
A debugging session is active.

	Inferior 1 [process 1348559] will be killed.

Quit anyway? (y or n) y
root@cluttered:~# 

```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde