ros2/kilted_tutorial_party #78: Check ROS 2 security CLI on Linux

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Linux.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-04:
Testing this ticket

danielcranston, 2025-05-06:
The link in the PR description goes to `master`, but it should probably go to `rolling`.

## Setup

```bash
$ dpkg --print-architecture 
amd64
$ lsb_release -d | grep Ubuntu
Description:	Ubuntu 24.04.2 LTS
source /opt/ros/kilted/setup.bash 
```

I went with the "**Install from debian packages**" route, so I haven't tested "**Install from source**". I also used CycloneDDS, since this issue requires it, but also because it's the only listed RMW that didn't require extra steps and just works out of the box 😅 

## Checks

- [x] **The Linux tutorial works**

Some comments sorted by section:

#### 1: Run the demo

Setting `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`, the talker/listener works as expected:

```bash
# Terminal 1
$ ros2 run demo_nodes_cpp talker --ros-args -e /talker_listener/talker
[INFO] [1746516603.745360962] [rcl]: Found security directory: /home/daniel/sros2_demo/demo_keystore/enclaves/talker_listener/talker
[INFO] [1746516604.772348871] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746516605.772313945] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746516606.772366616] [talker]: Publishing: 'Hello World: 3'
...
```

```bash
# Terminal 2
$ ros2 run demo_nodes_py listener --ros-args --enclave /talker_listener/listener 
[INFO] [1746515739.401083664] [rcl]: Found security directory: /home/daniel/sros2_demo/demo_keystore/enclaves/talker_listener/listener
[INFO] [1746515739.786703530] [listener]: I heard: [Hello World: 17]
[INFO] [1746515740.715905028] [listener]: I heard: [Hello World: 18]
[INFO] [1746515741.716532025] [listener]: I heard: [Hello World: 19]
...
```

The stuff about doing `export ROS_SECURITY_ENCLAVE_OVERRIDE=/talker_listener/listener` and testing CLI tools also worked.

#### 2: Run the demo _on different machines_

While I do have two laptops (amd64 and arm64), I can't get them to ping each other right now (just WSL things), so **I have not tested this part**! 

#### 3: Access Control

At this passage:

> The nodes will not be able to publish/subscribe to topics not listed in the policy. For example, the following attempt for the `listener` node to subscribe to a topic other than `chatter` will fail:
> 
> ```bash
> # This will fail because the node is not permitted to subscribe to topics other than chatter.
> ros2 run demo_nodes_py listener --ros-args -r chatter:=not_chatter -e /talker_listener/listener
> ```

Indeed it fails, but the error is very cryptic:

```
[...]
rclpy._rclpy_pybind11.RCLError: Failed to create subscription: failed to create topic [rt/not_chatter] for unknown reasons, at ./src/rmw_node.cpp:1976, at ./src/rcl/subscription.c:116
[ros2run]: Process exited with failure 1
```

#### 4: Certificate Revocation Lists

At the very end at "_Here you'll notice that the listener is not getting any data._", indeed the listener no longer receives data, and the talker is spamming this:

```
1746516955.671485 [0]        fsm: Begin handshake reply failed: Certificate not valid: error: certificate revoked; subject: /CN=\/talker_listener\/listener(code: 200)
```

So I think that's the intended behavior.

danielcranston, 2025-05-12:
I believe this can be closed.

closed 2025-05-12 by ahcorde