ros2/kilted_tutorial_party #79: Check ROS 2 security CLI on Linux

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Linux.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-08:
The PR description link takes you to the `master` branch, but should probably go to `kilted`.

Setup:

```bash
$ dpkg --print-architecture 
amd64
$ lsb_release -d | grep Ubuntu
Description:	Ubuntu 24.04.2 LTS
source /opt/ros/kilted/setup.bash 
```

### Checks

- [x] **The Linux tutorials work**

Some comments:

#### Debian FastDDS does not come with -DSECURITY=ON

```bash
$ ros2 run demo_nodes_cpp talker --ros-args --enclave /talker_listener/talker
[INFO] [1746520699.866077235] [rcl]: Found security directory: /home/daniel/sros2_demo/demo_keystore/enclaves/talker_listener/talker
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl node: This Fast DDS version doesn't have the security libraries
Please compile Fast DDS using the -DSECURITY=ON CMake option, at ./src/participant.cpp:406, at ./src/rcl/node.c:249
[ros2run]: Aborted
```

Even though the README suggests it would, since only the "**Install from source**" section mentions this.

So went with putting fastdds in my local workspace and rebuilding it. It was a bit unclear exactly what sources are required (`eProsima/Fast-DDS.git` seemed to be not enough, it requires other things as well), I ended up cloning the entire of

```bash
$ cd ~/colcon/ws/src
$ mkdir ros2_sources && cd ros2_sources
$ vcs import --input https://raw.githubusercontent.com/ros2/ros2/kilted/ros2.repos .
$ cd ~/colcon_ws/src

$ colcon build --cmake-args -DSECURITY=ON --packages-up-to fastdds
```

But that didn't turn out so great:

```
$ ros2 run demo_nodes_cpp talker 
[ros2run]: Segmentation fault
```

I'll try to look into it more later.

akky20, 2025-05-08:
@danielcranston , I also got the same error yesterday when I tried ..
DId you find any solution ?

danielcranston, 2025-05-08:
I found _a_ way to get this to work with FastDDS while still keeping (=not rebuilding from source) the majority of packages is `/opt/ros/kilted`:

I realized that the "**Install from source**" section says to build

```
colcon build --symlink-install --cmake-args -DSECURITY=ON --packages-select fastrtps rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp
```

(First of all, `fastrtps` was renamed `fastdds`, but the `rmw_` packages retains the "fastrtps" name, which is a bit confusing to me)

I noticed `fastdds` does not depend on `rmw_fastrtps_cpp`, it's the other way around! So I guess for good measure we ought to rebuild everything all the way up to `rmw_fastrtps_cpp`. After doing that,

```
colcon build --cmake-args -DSECURITY=ON --packages-up-to rmw_fastrtps_cpp
```

Then the segfault disappeared.

```
$ source ~/colcon_ws/install/setup.bash
$ export ROS_SECURITY_KEYSTORE=~/sros2_demo/demo_keystore
$ export ROS_SECURITY_ENABLE=true
$ export ROS_SECURITY_STRATEGY=Enforce
$ ros2 run demo_nodes_cpp talker
```

But at this point a large number of packages (101!) are rebuilt in the local workspace and shadowing the `/opt/ros/kilted` ones, so it's kind of a hybrid debian/source situation... I'll try again without `--packages-up-to` but rather enumerating the important ones listed in the "**Install from source**" section.

@tfoote @cottsay @mjcarroll and maybe others, do you know if it's intended that Kilted debians come with a version of FastDDS **_that was not build with -DSECURITY=ON_** ? If it's intended, then the README definitely needs to be updated to instruct the user to rebuild FastDDS from source in both cases.

danielcranston, 2025-05-08:
> I'll try again without `--packages-up-to` but rather `--packages-select` on the important ones listed in the "**Install from source**" section.

Well that did not work. First of all a few more packages needed to be listed for the build to even kick off. Once it did, `rmw_fastrtps_shared_cpp` failed to build:

```
$ rm -rf install/ build/ log/
$ colcon build --cmake-args -DSECURITY=ON --packages-select fastdds rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp fastcdr foonathan_memory_vendor gtest_vendor

[...]

[2.918s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /home/daniel/colcon_ws/install/google_benchmark_vendor/share/google_benchmark_vendor/package.sh
- /home/daniel/colcon_ws/install/osrf_testing_tools_cpp/share/osrf_testing_tools_cpp/package.sh
- /home/daniel/colcon_ws/install/gmock_vendor/share/gmock_vendor/package.sh
- /home/daniel/colcon_ws/install/test_interface_files/share/test_interface_files/package.sh
- /home/daniel/colcon_ws/install/ament_cmake_google_benchmark/share/ament_cmake_google_benchmark/package.sh
- /home/daniel/colcon_ws/install/ament_cmake_vendor_package/share/ament_cmake_vendor_package/package.sh
- /home/daniel/colcon_ws/install/ament_mypy/share/ament_mypy/package.sh
- /home/daniel/colcon_ws/install/ament_pycodestyle/share/ament_pycodestyle/package.sh
- /home/daniel/colcon_ws/install/ament_cmake_mypy/share/ament_cmake_mypy/package.sh
- /home/daniel/colcon_ws/install/mimick_vendor/share/mimick_vendor/package.sh
- /home/daniel/colcon_ws/install/performance_test_fixture/share/performance_test_fixture/package.sh
Failed   <<< rmw_fastrtps_shared_cpp [0.03s, exited with code 1]
```

So the  `--packages-up-to` approach in the previous comment is still the best I've got. It works, but it's definitely not what's written in the instructions.

Aniketkumarroy, 2025-05-08:
Hey I will be testing this

cottsay, 2025-05-08:
This was indeed a bug in the Rolling and Kilted debs for the `fastdds` package. I've pushed a fix, we're just waiting on the buildfarm to rebuild everything.

Aniketkumarroy, 2025-05-09:
Hi I tested this feature and it is working
### **Assigning enclave to node**

[image]

### **overriding the enclave**

[image]

everything is working fine

danielcranston, 2025-05-10:
Indeed it seems good now! I tried a couple of hours after @cottsay's message, but I guess the debs had not been published yet. Trying again now I see some new versions:

```
$ sudo apt update
$ apt list --upgradable | grep -E "ros-kilted-fastdds|ros-kilted-rmw-fastrtps-cpp"
ros-kilted-fastdds/noble 3.2.2-1noble.20250508.201326 amd64 [upgradable from: 3.2.1-2noble.20250424.202832]
ros-kilted-rmw-fastrtps-cpp/noble 9.3.2-2noble.20250508.204011 amd64 [upgradable from: 9.3.2-2noble.20250424.212103]
```

And indeed it seems FastDDS related packages were built `-DSECURITY=ON` this time around.

- [x] **The Linux tutorials work**
  - [x] Talker/listener works as intended
  - [x] Using CLI with ROS_SECURITY_ENCLAVE_OVERRIDE
  - [ ] Running the demo on different machines - **_not tested_**
  - [x] Access control
```
$ ros2 run demo_nodes_py listener --ros-args -r chatter:=not_chatter -e /talker_listener/listener
[INFO] [1746845065.469438200] [rcl]: Found security directory: /home/daniel/sros2_demo2/demo_keystore/enclaves/talker_listener/listener
2025-05-10 11:44:25.924 [SECURITY Error] rt/not_chatter topic not found in allow rule. (./src/cpp/security/accesscontrol/Permissions.cpp:1247) -> Function check_create_datareader
2025-05-10 11:44:25.924 [SECURITY Error] Error checking creation of local reader  (rt/not_chatter topic not found in allow rule. (./src/cpp/security/accesscontrol/Permissions.cpp:1247))
 -> Function get_datareader_sec_attributes
2025-05-10 11:44:25.924 [DATA_READER Error] Problem creating associated Reader -> Function enable
Traceback (most recent call last):
  File "/opt/ros/kilted/lib/demo_nodes_py/listener", line 33, in <module>
    sys.exit(load_entry_point('demo-nodes-py==0.36.0', 'console_scripts', 'listener')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/kilted/lib/python3.12/site-packages/demo_nodes_py/topics/listener.py", line 35, in main
    node = Listener()
           ^^^^^^^^^^
  File "/opt/ros/kilted/lib/python3.12/site-packages/demo_nodes_py/topics/listener.py", line 26, in __init__
    self.sub = self.create_subscription(String, 'chatter', self.chatter_callback, 10)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/kilted/lib/python3.12/site-packages/rclpy/node.py", line 1733, in create_subscription
    subscription_object = _rclpy.Subscription(
                          ^^^^^^^^^^^^^^^^^^^^
rclpy._rclpy_pybind11.RCLError: Failed to create subscription: create_datareader() could not create data reader, at ./src/subscription.cpp:686, at ./src/rcl/subscription.c:116
[ros2run]: Process exited with failure 1
```
  - [x] Certificate Revocation Lists
Small note: neither talker nor listener log any errors, which is different from when using CycloneDDS (https://github.com/ros2/kilted_tutorial_party/issues/78), where the talker would log "Begin handshake reply failed [...]".

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde