ros2/kilted_tutorial_party #8: Beginner CLI Tools

tfoote, 2025-05-01:
Check the documentation for the 'Beginner CLI Tools' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Beginner CLI Tools page](https://docs.ros.org/en/kilted/Tutorials/Beginner-CLI-Tools.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-01:
please assign this to me.

khaledgabr77, 2025-05-01:
# Configuring environment

[image]

# Using turtlesim, ros2, and rqt

[image]

[image]

[image]

# Understanding nodes

[image]

[image]

# Understanding topics

[image]

[image]

[image]

[image]

# Understanding services

[image]

[image]

[image]

[image]

[image]

[image]

# Understanding parameters

[image]

[image]

# Understanding actions

[image]

# Using rqt_console to view logs

[image]

# Launching nodes

[image]

# Recording and playing back data

[image]

closed 2025-05-09 by mjcarroll