ros2/kilted_tutorial_party #80: Check ROS 2 security CLI on Linux

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Linux.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-07:
Working on this !!

akky20, 2025-05-07:
Setting Key
[image]

running the Zenoh Router
[image]

It founds the security directory
[image]

[image]

Let me know if anything needed to do further

ahcorde, 2025-05-07:
Thank you for testing