ros2/kilted_tutorial_party #81: Check ROS 2 security CLI on Windows

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Windows.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / Pixi Kilted Env / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Testing Issue #81 (Security Tutorial) on Windows 11 / amd64 / Pixi Kilted Env / CycloneDDS (`rmw_cyclonedds_cpp`).

**Result: FAILED**

- The initial steps (creating keystore, enclaves, getting policies, creating permissions) completed without error.
- Environment variables were set correctly (`ROS_SECURITY_ENABLE=true`, `ROS_SECURITY_STRATEGY=Enforce`, `ROS_SECURITY_KEYSTORE`, `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`).
- However, **all attempts** to run nodes (`ros2 run demo_nodes_py talker/listener` or `ros2 run demo_nodes_cpp talker/listener`) with security enabled using CycloneDDS failed during node initialization.

**Error Analysis:**
The consistent error indicates that `rmw_cyclonedds_cpp` fails to create the DDS participant because it cannot find a required certificate file:
`Error occurred while validating local permissions: load_file_into_BIO: File could not be found, opened or is empty, path: .../enclaves/.../permissions_ca.cert.pem`
This `permissions_ca.cert.pem` file is not generated by the standard `ros2 security create_enclave` or `create_permission` commands.

**Conclusion:** The SROS2 tutorial workflow appears incompatible with CycloneDDS on the Kilted Windows binary setup due to missing security artifact requirements specific to CycloneDDS's security plugins. The nodes fail to start when security is enforced with CycloneDDS.

**Terminal Output Snippet (Talker Failure):**
<details>
<summary>Click to expand</summary>
(pixi_ros2_rolling) C:\Users\jztsh\Documents\sros2_demo>ros2 run demo_nodes_py talker --ros-args --enclave /talker_listener/talker
[INFO] [1746362734.637211300] [rcl]: Found security directory: C:\Users\jztsh\Documents\sros2_demo\demo_keystore\enclaves\talker_listener\talker
1746362746.224205 [0] 1672: Error occurred while validating local permissions: load_file_into_BIO: File could not be found, opened or is empty, path: C:/Users/jztsh/Documents/sros2_demo/demo_keystore/enclaves/talker_listener/talker/permissions_ca.cert.pem(code: 117)
[ERROR] [1746362746.225881500] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create DDS participant
C:\pixi_ws\ros2-windows\Lib\site-packages\rclpy\node.py:211: RuntimeWarning: Failed to fini rosout publisher: rcl node implementation is invalid, at C:\ci\ws\src\ros2\rcl\rcl\src\rcl\node.c:387
self.__node = _rclpy.Node(
Traceback (most recent call last):
File "\?\C:\pixi_ws\ros2-windows\Lib\demo_nodes_py\talker-script.py", line 33, in <module>
sys.exit(load_entry_point('demo-nodes-py==0.36.0', 'console_scripts', 'talker')())
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\pixi_ws\ros2-windows\Lib\site-packages\demo_nodes_py\topics\talker.py", line 42, in main
node = Talker()
^^^^^^^^
File "C:\pixi_ws\ros2-windows\Lib\site-packages\demo_nodes_py\topics\talker.py", line 25, in init
super().init('talker')
File "C:\pixi_ws\ros2-windows\Lib\site-packages\rclpy\node.py", line 211, in init
self.__node = _rclpy.Node(
^^^^^^^^^^^^
rclpy._rclpy_pybind11.RCLError: error creating node: error not set, at C:\ci\ws\src\ros2\rcl\rcl\src\rcl\node.c:250
[ros2run]: Process exited with failure 1
</details>

It won't work the tutorial itself mentions for FastDDS only, but still tested.

Aniketkumarroy, 2025-05-13:
Hi I tested this for my setup and it worked, I am using Windows 11
### ***Setup***

[image]
### ***Running Authentication and Encryption***

[image]
### ***Access Control***
using policies/sample.policy.xml

[image]
running nodes

[image]

- [x] The Linux tutorials work

JatinVira, 2025-05-14:
@ahcorde This can be closed if it suffices the purpose

ahcorde, 2025-05-14:
Thank you for testing

closed 2025-05-14 by ahcorde