ros2/kilted_tutorial_party #82: Check ROS 2 security CLI on Windows

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Tutorial URL](https://github.com/ros2/sros2/blob/master/SROS2_Windows.md)

## Checks
- [ ] **The Linux tutorials work**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this (Security Tutorial) on Windows 11 / amd64 / binary / Fast DDS

ShahazadAbdulla, 2025-05-04:
Successfully completed all checks for Issue #82 (Security Tutorial) on Windows 11 / amd64 / Pixi Kilted Env / Fast DDS.

- **Prerequisites:** Installed `svn` via Chocolatey successfully.
- **Keystore/Enclave Generation:** `create_keystore` and `create_enclave` commands worked as expected (handled `.rnd` file by setting environment variable).
- **Environment Variables:** Security variables (`ROS_SECURITY_KEYSTORE`, `ROS_SECURITY_ENABLE`, `ROS_SECURITY_STRATEGY`) set correctly in multiple terminals relative to the `sros2_demo` directory.
- **Auth/Encrypt Test:** Talker and Listener nodes communicated successfully using security and specified enclaves.
- **Policy/Permission Generation:** Successfully obtained policy files (by cloning `sros2` repo as `svn checkout` failed) and generated permissions using `create_permission`.
- **Access Control (Allowed):** Talker and Listener communicated successfully using permissions and enclaves.
- **Access Control (Denied):** Attempting to run the listener remapped to `/not_chatter` correctly failed with security errors (`topic not found in allow rule`, `Failed to create subscription`), preventing communication as expected by the policy.

All stages of the security tutorial were validated successfully on this setup.

closed 2025-05-05 by ahcorde