ros2/kilted_tutorial_party #89: Check ROS 2 topic CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check bw**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic bw /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       Subscribed to [/joint_states]
       5.04 KB/s from 50 messages
         Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
       5.02 KB/s from 100 messages
         Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
       ...
       ```
  </details>
- [ ] **Check delay**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic delay /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       average delay: 0.001
         min: 0.000s max: 0.001s std dev: 0.00012s window: 47
       average delay: 0.001
         min: 0.000s max: 0.002s std dev: 0.00021s window: 97
       ...
       ```
  </details>
- [ ] **Check echo**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       ---
       header:
         stamp:
           sec: 1649125169
           nanosec: 961952254
         frame_id: ''
       name:
       - single_rrbot_joint1
       - single_rrbot_joint2
       position:
       - -0.9985924716759744
       - -0.9985924716759744
       velocity: []
       effort: []
       ---
       ...
       ```
  </details>
- [ ] **Check find**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic find sensor_msgs/msg/JointState
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /joint_states
       ```
  </details>
- [ ] **Check hz**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic hz /joint_states
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       average rate: 50.252
         min: 0.000s max: 0.043s std dev: 0.00451s window: 52
       average rate: 50.125
         min: 0.000s max: 0.043s std dev: 0.00492s window: 102
       ...
       ```
  </details>
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic info /joint_states
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Type: sensor_msgs/msg/JointState
       Publisher count: 1
       Subscription count: 1
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /joint_states
       /map
       /parameter_events
       /robot_description
       /rosout
       /scan
       /tf
       /tf_static
       ```
  </details>
- [ ] **Check type**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic type /joint_states
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       sensor_msgs/msg/JointState
       ```
  </details>
- [ ] **Check pub**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp listener
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic pub /chatter std_msgs/msg/String "{data: 'hi'}"
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 1
       [INFO] [1649125551.479505602] [listener]: I heard: [hi]
       [INFO] [1649125552.480104544] [listener]: I heard: [hi]
       ...
       ```
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       publisher: beginning loop
       publishing #1: std_msgs.msg.String(data='hi')
       
       publishing #2: std_msgs.msg.String(data='hi')
       ...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
I am working on this

suchetanrs, 2025-05-02:
The output of all the commands is below.

The file used to test can be found here https://github.com/suchetanrs/kilted_tutorial_party/tree/main/scripts

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ./scripts/ros2_
ros2_help.sh       ros2_topic_cli.sh  ros2_wtf_cli.sh    
root@cluttered:~# ./scripts/ros2_topic_cli.sh 
==================ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-03-01-14-26-641086-cluttered-1323417
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [1323474]
[INFO] [robot_state_publisher-2]: process started with pid [1323475]
[INFO] [dummy_joint_states-3]: process started with pid [1323476]
[INFO] [dummy_laser-4]: process started with pid [1323477]
[dummy_laser-4] [INFO] [1746215066.949128816] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1746215066.949314525] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1746215066.949352455] [dummy_laser]: scan time increment: 	0.000000
[robot_state_publisher-2] [INFO] [1746215066.960072495] [robot_state_publisher]: Robot initialized
==================ros2 topic bw /joint_states
Subscribed to [/joint_states]
5.02 KB/s from 50 messages
	Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
5.01 KB/s from 100 messages
	Message size mean: 0.10 KB min: 0.10 KB max: 0.10 KB
==================ros2 topic delay /joint_states
average delay: 0.001
	min: 0.001s max: 0.001s std dev: 0.00013s window: 49
average delay: 0.001
	min: 0.001s max: 0.001s std dev: 0.00013s window: 99
average delay: 0.001
	min: 0.001s max: 0.001s std dev: 0.00011s window: 149
==================ros2 topic echo --once /joint_states
header:
  stamp:
    sec: 1746215368
    nanosec: 612735522
  frame_id: ''
name:
- single_rrbot_joint1
- single_rrbot_joint2
position:
- -0.26323179136582836
- -0.26323179136582836
velocity: []
effort: []
---              
==================ros2 topic find sensor_msgs/msg/JointState
/joint_states
==================ros2 topic hz /joint_states
average rate: 49.994
	min: 0.020s max: 0.020s std dev: 0.00012s window: 49
average rate: 50.000
	min: 0.020s max: 0.020s std dev: 0.00010s window: 99
average rate: 50.000
	min: 0.020s max: 0.020s std dev: 0.00010s window: 149
==================ros2 topic info /joint_states
Type: sensor_msgs/msg/JointState
Publisher count: 1
Subscription count: 1
==================ros2 topic list
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
==================ros2 topic type /joint_states
sensor_msgs/msg/JointState
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='hi')

[INFO] [1746215095.040925219] [listener]: I heard: [hi]
publishing #2: std_msgs.msg.String(data='hi')

[INFO] [1746215096.041942977] [listener]: I heard: [hi]
publishing #3: std_msgs.msg.String(data='hi')

[INFO] [1746215097.042005087] [listener]: I heard: [hi]
publishing #4: std_msgs.msg.String(data='hi')

[INFO] [1746215098.041976536] [listener]: I heard: [hi]
publishing #5: std_msgs.msg.String(data='hi')

[INFO] [1746215099.041885171] [listener]: I heard: [hi]
publishing #6: std_msgs.msg.String(data='hi')

[INFO] [1746215100.041980398] [listener]: I heard: [hi]
publishing #7: std_msgs.msg.String(data='hi')

[INFO] [1746215101.041876954] [listener]: I heard: [hi]
^C[INFO] [1746215101.582975061] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746215101.583311777] [rclcpp]: signal_handler(signum=2)
[dummy_laser-4] [INFO] [1746215101.582950938] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1746215101.582996290] [rclcpp]: signal_handler(signum=2)
[dummy_joint_states-3] [INFO] [1746215101.582986720] [rclcpp]: signal_handler(signum=2)
[dummy_map_server-1] [INFO] [1746215101.582990243] [rclcpp]: signal_handler(signum=2)
[INFO] [dummy_laser-4]: process has finished cleanly [pid 1323477]
[INFO] [dummy_map_server-1]: process has finished cleanly [pid 1323474]
[INFO] [dummy_joint_states-3]: process has finished cleanly [pid 1323476]
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 1323475]
root@cluttered:~# 
```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde