ros2/kilted_tutorial_party #93: Check ROS 2 wtf CLI

tfoote, 2025-05-01:
`ros2 wtf` is an alias for `ros2 doctor`.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Testing Issue #93 on Windows 11 / amd64 / binary / CycloneDDS.

Confirmed that `ros2 wtf` functions as an alias for `ros2 doctor` and produces identical output and behavior.

**Results:**
*   `ros2 wtf -r`: Partially failed, produced the same `No module named 'rosdistro'` warnings and missing report sections as `ros2 doctor -r` (#38).
*   `ros2 wtf -rf`: Failed with `1/3 checks failed`, showing the same network warnings and missing `rosdistro` dependency errors as `ros2 doctor -rf` (#38).
*   `ros2 wtf -iw`: Failed with `1/3 checks failed`, showing the same warnings/errors as `ros2 doctor -iw` (#38).
*   `ros2 wtf hello`: Passed, successfully printed the repeating multicast summary messages, identical to `ros2 doctor hello` (#38).

**Conclusion:** The `ros2 wtf` alias works, but exhibits the same underlying failures related to the missing `rosdistro` dependency and network checks on Windows binary as observed in #38.

dilpreetsingh25, 2025-05-15:
#### Testing on Window 11.

### ros2 wtf -r 

- [x] Check -r 

####  Installed rosdistro
#### Command Successful 
```
(pixi_ros2_kilted) C:\pixi_ws>ros2 wtf -r
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.

   PACKAGE VERSIONS
actionlib_msgs                            : latest=5.5.0, local=5.5.0
action_msgs                               : latest=2.3.0, local=2.3.0
action_tutorials_cpp                      : latest=0.36.0, local=0.36.0
action_tutorials_py                       : latest=0.36.0, local=0.36.0
ament_clang_format                        : latest=0.19.2, local=0.19.2
ament_clang_tidy                          : latest=0.19.2, local=0.19.2
ament_cmake                               : latest=2.7.3, local=2.7.3
ament_cmake_auto                          : latest=2.7.3, local=2.7.3
'
'
'
'
visualization_msgs                        : latest=5.5.0, local=5.5.0
yaml_cpp_vendor                           : latest=9.1.0, local=9.1.0
zenoh_cpp_vendor                          : latest=0.6.0, local=0.6.0
zenoh_security_tools                      : latest=0.6.0, local=0.6.0
zstd_vendor                               : latest=0.32.0, local=0.32.0

   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-11-10.0.26100-SP0
release          : 11
processor        : Intel64 Family 6 Model 183 Stepping 1, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0

(pixi_ros2_kilted) C:\pixi_ws>
```

### ros2 wtf -rf
- [ ] Check -rf
#### Command Failed 
```
(pixi_ros2_kilted) C:\pixi_ws>ros2 wtf -rf
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 87: UserWarning: ERROR: No loopback IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 93: UserWarning: No multicast IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\package.py: 122: UserWarning: point_cloud_transport has been updated to a new version. local: 5.1.2 < latest: 5.1.3
'
'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\package.py: 130: UserWarning: Cannot find the latest versions of packages: rmw_connextddsmicro rosidl_generator_tests rosidl_typesupport_introspection_tests [...]. Use `ros2 doctor --report` to see full list.

1/5 check(s) failed

Failed modules: network
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
```
### ros2 wtf -iw 
- [ ] Check -iw

#### Command Failed
```
(pixi_ros2_kilted) C:\pixi_ws>ros2 wtf -iw
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 87: UserWarning: ERROR: No loopback IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 93: UserWarning: No multicast IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\package.py: 122: UserWarning: point_cloud_transport has been updated to a new version. local: 5.1.2 < latest: 5.1.3
'
'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\package.py: 122: UserWarning: rviz_visual_testing_framework has been updated to a new version. local: 15.0.1 < latest: 15.0.2
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\package.py: 130: UserWarning: Cannot find the latest versions of packages: rmw_connextddsmicro rosidl_generator_tests rosidl_typesupport_introspection_tests [...]. Use `ros2 doctor --report` to see full list.

2/5 check(s) failed

Failed modules: package network

(pixi_ros2_kilted) C:\pixi_ws># User input in terminal 1
'#' is not recognized as an internal or external command,
operable program or batch file.
```

### ros2 wtf hello 
- [x] Check -hello
#### Command Successful
```
(pixi_ros2_kilted) C:\pixi_ws>ros2 wtf hello
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 11
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
```