ros2/kilted_tutorial_party #94: Check ROS 2 wtf CLI

tfoote, 2025-05-01:
`ros2 wtf` is an alias for `ros2 doctor`.

## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 wtf hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
- [x] Check -r

Large report generated with the ending info:
```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-58-generic-x86_64-with-glibc2.39
release          : 6.8.0-58-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```

- [x] Check -rf

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ros2 wtf -rf

All 5 checks passed

root@cluttered:~# 
```

- [x] Check -iw

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# # User input in terminal 1
ros2 wtf -iw

All 5 checks passed

root@cluttered:~# 
```

- [x] Check hello

```
root@cluttered:~# source /opt/ros/kilted/setup.bash                                                                                     [0/0]
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp         
root@cluttered:~# # User input in terminal 1
ros2 wtf -iw

All 5 checks passed

root@cluttered:~# # User input in terminal 1
ros2 wtf hello
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 11
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 11
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde