ros2/kilted_tutorial_party #99: Colcon Tutorial

tfoote, 2025-05-01:
Check the documentation for the 'Colcon Tutorial' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Colcon Tutorial page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Aniketkumarroy, 2025-05-13:
Hi, I too tested on my setup and here are the terminal traces

[image]

[image]

[image]

[image]

[image]
### ***tab completion***

https://github.com/user-attachments/assets/f8dcdb83-9d7e-4703-9d78-6835f730f41d
### ***My System***

[image]

all things work fine, @ahcorde I think we can close this issue now

ahcorde, 2025-05-13:
Thank you for testing, Related PR is merged

closed 2025-05-13 by ahcorde