ros2/kilted_tutorial_party #568: Installation instructions

mjcarroll, 2025-05-04:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Ubuntu Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-04:
installing via source, will confirm in few mins

dheerubhai-101, 2025-05-04:
- [x] Install works

Install works, but just saw some deprecation warnings and errors with VCS

```
--- stderr: gz_math_vendor                                                                                                                                         
/usr/bin/vcs:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
```
Also some additional warnings in the form `stderr`, of `UserWarning: Unbuilt egg for pytest-repeat`

[25.334s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'rosbag2_performance_benchmarking_msgs' because it has no 'install' target

But everything built successfully

```
Summary: 370 packages finished [11min 44s]
  97 packages had stderr output: action_tutorials_py ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint camera_info_manager_py demo_nodes_py domain_coordinator examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher examples_tf2_py launch launch_pytest launch_ros launch_testing launch_testing_examples launch_testing_ros launch_xml launch_yaml lifecycle_py osrf_pycommon qt_dotgraph quality_of_service_demo_py ros2action ros2bag ros2cli ros2component ros2doctor ros2interface ros2launch ros2lifecycle ros2multicast ros2node ros2param ros2pkg ros2run ros2service ros2test ros2topic ros2trace rosbag2_examples_py rosbag2_performance_benchmarking rosbag2_py rosidl_cli rosidl_pycommon rosidl_runtime_py rpyutils rqt rqt_action rqt_bag rqt_bag_plugins rqt_console rqt_graph rqt_gui rqt_gui_py rqt_msg rqt_plot rqt_publisher rqt_py_console rqt_reconfigure rqt_service_caller rqt_shell rqt_srv rqt_topic rviz_common sensor_msgs_py sros2 test_launch_ros test_ros2trace test_tracetools_launch tf2_ros_py tf2_tools topic_monitor tracetools_launch tracetools_read tracetools_test tracetools_trace 
```

[image]

closed 2025-05-05 by mjcarroll